exelban/stats #854: Nothing happen after launching the app

Ricardokevins, 2022-02-22:
Install from brew or stats.dmg ( Newest Version )
successfully Start the app.
And The top bar does not display the corresponding information.
I use MacBookPro M1pro . MacOS 12.2.1

Thank you for any suggestion for solving problem!

orchowski, 2022-02-22:
I heard that there are some problems after the most recent update. On the same device, MacOS 12.2 works well for me

exelban, 2022-02-22:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in `Applications -> Utilities -> Console`. Find the `Crash Reports` tab. There you must see a row called Stats.

Ricardokevins, 2022-02-22:
> Hi. Thanks for reporting. Could you please share the crash report? It could be found in `Applications -> Utilities -> Console`. Find the `Crash Reports` tab. There you must see a row called Stats.

Thank you for your quick reply.
I think it's some problems with the display mechanism of the top bar.

The app is running normally (not crash because i can see the setting UI and no crash report observed as well )

The camera in the middle of the top bar takes up a lot of space which obstructs the normal display. ( I also encounter some problem like this : Icons of some apps disappear or looks strange)

After removing some app which will display in top bar (get more space ), it seems that this app returns to normal

Ricardokevins, 2022-02-22:
> I heard that there are some problems after the most recent update. On the same device, MacOS 12.2 works well for me

The camera in middle of top bar makes it uncomfortable to display the app's information.
The displayable space is tighter..... I think maybe the reason for my problem.
Removing unnecessary app to get more displayable space solves the problem

closed 2022-02-22 by Ricardokevins