exelban/stats #859: Bluetooth: battery percent not showing

lcardito, 2022-02-28:
Hey team! 

Just found out that the mac bluetooth menu now displays the battery of my logic mouse and my Jabra headphones when connected. `stats` seems to be picking those up however the widget stays blank, not sure if those external devices are just not supported, I am doing something wrong or there's a bug somewhere?

OS: macOS Monterey (12.2.1)
Stats: Version 2.7.5

Attaching some screenshot for clarity

[image]

[image]

Many thanks!

exelban, 2022-03-03:
Hi. Thanks for the feedback.
There is a file `/Library/Preferences/com.apple.Bluetooth`. Could you open it and find the `DeviceCache` field. Could you please post the data inside this field? PS: omit sensitive data.

lcardito, 2022-03-04:
Hey @exelban ! Thanks for the response!

Looking in that folder I don't have a `com.apple.Bluetooth` file I am afraid.. only the binary `plist` there 😞 

```
❯ ll /Library/Preferences/ | grep Bluetooth
-rw-r--r--   1 root         wheel          39K  2 Nov 15:00 com.apple.Bluetooth.plist
```

exelban, 2022-03-04:
Yeah, it's exactly that file. `plist` is not a binary, it's a text file.

lcardito, 2022-03-07:
Hey, thanks again and sorry for the slow response loop :)

I am not familiar with this file but I think it's not exactly readable.. 

```
DeviceCache_PersistentPortsServices_ControllerPowerState_BluetoothVersionNumber�	�       ! " # V ` k p z �_
```

Wonder if I should try and delete this `plist`!? 🤔

exelban, 2022-03-07:
Yes, it's exactly this file. Please open it with any text editor and find the `DeviceCache` part.

lcardito, 2022-03-07:
Not sure if this helps?

```
HIDDevicesNeedsSDP[DeviceCache_PersistentPortsServices_ControllerPowerState_BluetoothVersionNumber�	�       ! " # V ` k p z �_$29955860-9E29-4CE8-8F0E-78DC8C0F8FB2_$44D92A02-DDD5-4F1B-A70D-CDB78B522353_$0C3E4477-D6F1-4243-9EC3-B7C6AC589CC9_$52A25118-39AA-4E2D-8019-7E74F04FA115_$ADBF40FD-F346-4210-8704-357626E31749_$44713FFD-C0E7-40D4-B440-25A9AEB67237_$3EAFEECB-832C-4138-A121-65893B29D998� $ % & ' ( ) * + , - . / 0 1 2 3 4 5 6 7 8 9 : ; < = > ? @ A B C C E F G B H I J K L M I N O P Q R S T I  IXServices^LastNameUpdate]ClassOfDevice_PageScanRepetitionMode\FirstPairing_DeviceAddressType_LastInquiryUpdate\ManufacturerTName^VendorIDSource]DeviceAddress\PageScanMode_EncryptionKeySize]LMPSubversion_SupportedFeaturesWEIRData[ClockOffsetZLMPVersion[ServiceMaskXVendorID_SCOConnectionInterval_ExtendedFeaturesPage1YProductID_LastServicesUpdate^Page
```

exelban, 2022-03-07:
Or you can use any online tool to read the file.
PS: I was thinking that any text editor can read this file, but not(

lcardito, 2022-03-15:
Hey - apologies for the late reply.

There doesn't seem to be a way of reading this file. Would it help if I delete it and try to start fresh?

izacus, 2022-03-15:
I have the same kind of issue - there are no Bluetooth devices listed:

[image]

Funny enough, there's no `com.apple.Bluetooth` at all in /Library/Preferences. This is on a M1 16" MBP with Monterey 12.3.

webjib, 2022-04-05:
Just for your information, macOS Monterey does not use the com.apple.Bluetooth.plist anymore. Instead, there are two SQL database in /Library/Bluetooth

exelban, 2022-04-06:
Hi. Yes, starting from macOS 12.3 there is no `com.apple.Bluetooth`. But looks like macOS started reporting the battery levels to the `system_profiler`. I will add a reader for this. This week will be some MVP.

closed 2022-04-10 by exelban