exelban/stats #863: [Feature request] Show CPU distribution between performance/efficiency cores

tillmon, 2022-03-02:
Would be interesting to see the CPU distribution between performance/efficiency cores of M1 chips, similar as App Tamer is doing it.

[Screenshot](https://www.stclairsoft.com/AppTamer/images/MainPanelThumb.png)

exelban, 2022-03-03:
Hi. Thanks for the feedback.
It could be a nice feature.

macbugs, 2022-03-19:
note, this feature is currently supported by MenuMeters, https://github.com/exelban/stats/issues/565#issuecomment-1072996875

exelban, 2022-03-19:
@macbugs so what?

macbugs, 2022-03-20:
There are some bugs in various applications which can be observed by viewing the core load separately.  For example, a single core will stick at 100% utilization.  So, people who may need this feature immediately for troubleshooting purposes should use the other program.  Is it a crime to help others by sharing useful information?

zigapovhe, 2022-05-14:
> Would be interesting to see the CPU distribution between performance/efficiency cores of M1 chips, similar as App Tamer is doing it.
> 
> [Screenshot](https://www.stclairsoft.com/AppTamer/images/MainPanelThumb.png)

Darn, that would be cool! Awesome suggestion @tillmon . Big thanks to @exelban for developing and maintaining stats!  This is one of the coolest menu stats app for macOS out there (especially with M1 chip support)! Constantly maintained, sleek design, free and open source. Keep it up! 👏🏻🏆

exelban, 2022-08-14:
Hi. I have added information about efficiency and performance cores usage to the popup view.
Also, you can colorize the clusters in the bar chart widget. For now, I have no plans to add this info to the chart. Because it will require redesigning the chart. But I will do it someday)

I hope it will be ok for now.

closed 2022-08-14 by exelban