exelban/stats #865: Ability to choose "Based on utilization" for battery widget color

boldandbrad, 2022-03-05:
I like being able to set "based on utilization" for my GPU, CPU, and RAM menu bar icons colors because it offers another dimension to the status. I believe something could be done for battery as well. Since we have the ability to show battery percentage, perhaps 50% or more is blue, and lower battery levels change to orange or red - almost on the opposite scale the other metrics use for higher percentages. "based on stress" would be cool too if there are applications/processes currently utilizing a lot of battery power.

exelban, 2022-03-06:
Hi. Thanks for the feedback.
It could be added.

exelban, 2022-03-07:
Hi. I have checked and there is an `based on utilisation` color option in the Mini and Bar chart widget.

boldandbrad, 2022-03-09:
[image]

This option is not available to me. I'm on a 2020 Intel MacBook Pro running macOS 12.2.1

exelban, 2022-03-10:
Oh, sorry. My bad. I was noticed CPU, GPU, and RAM. And think that you asking about these modules(
Will be added.

closed 2022-03-13 by exelban