exelban/stats #866: Unable to see bluetooth list in settings

wencczhu, 2022-03-07:
macOS version: 12.2.1 
Stats version: 2.7.6

I can't see the list of paired bluetooth devices in the settings
[image]

This is what the status bar look like
[image]

I think I have granted bluetooth permission to the app
[image]

Btw, I don't see any issues with another mac (mini)

exelban, 2022-03-07:
Hi. Thanks for the feedback.
If you don't see any BLE device it could be because of:
- there is no paired devices
- there is no device's that  battery level could be obtained

Th0m4, 2022-03-08:
Hi, 
I have the same problem ! 
Did you know on wich devices the battery level can be obtained ?
I have Airpods 2, MX masted and MX keys paired and connected to my macbook but nothing in the settings

exelban, 2022-03-09:
Hi. This thread is a duplicate of #859 
Please continue there.

closed 2022-03-09 by exelban