exelban/stats #869: Network Widget settings values for color of download/color of upload are swapped

aetheode, 2022-03-12:
**Describe the bug**
[image]

I believe the first value should state the following, since they appear this way in the UI
Color of upload: second red
Color of down: second blue

**Details:**
 - Device: MacBook Pro (16-inch, 2021)
 - macOS: 12.2.1
 - Application version: 2.7.6

exelban, 2022-03-12:
Hi. Thanks for the feedback.
Maybe I have made a mistake) Will check that.

exelban, 2022-03-13:
Hi. The colors in the chart are correct.

aetheode, 2022-03-13:
Hi @exelban , how can that be? 

In the UI on the right, it says upload is "second red", but in the UI on the left, you can see it says download is "second red"

exelban, 2022-03-14:
Ohh. I see. You are talking about widget settings. Widget settings do not connect with popup view. These settings will change only widget colors.

valynor, 2022-04-04:
> Ohh. I see. You are talking about widget settings. Widget settings do not connect with popup view. These settings will change only widget colors.

The bug is still there - the descriptions in Network -> Widget Settings are switched! 

[image]

exelban, 2022-04-04:
Hi. Ok, now I see. My mistake. Will be fixed.

closed 2022-04-10 by exelban