exelban/stats #870: Endless spam messages - Console

johans3, 2022-03-12:
I got a problem with the **harddrive widget** 
The service itself does its job, but if activated, i get a sea of spam messages in Apple's **Console**, the line is always the same like this:
`standard	11:32:59.774099 +0100	diskarbitrationd	<private>`
the diskarbitrationd daemon monitors disks, usb interfaces and **Stat** tries to read some of its private informations as well

Stat version: 2.7.5
Mac: 10.14.6

exelban, 2022-03-12:
Hi. Thanks for the feedback.
Will check that.

exelban, 2022-03-12:
I can confirm that there are a lot of these logs in the console.
But according to [this documentation](https://developer.apple.com/library/archive/documentation/DriversKernelHardware/Conceptual/DiskArbitrationProgGuide/Introduction/Introduction.html). The Disk Arbitration it's a framework that also returns a information about a drive.
And the log hit every time Stats reads a space and I/O activity. There is nothing I can do with that.

closed 2022-03-20 by exelban