exelban/stats #871: Sensor stats label and reodering sensor reading

kprinssu, 2022-03-12:
**Describe the bug**

There is no option to combine the vertical label "SEN" with the sensors. Right now the options are two separate menu bar entries, can this be combined into one?

Please see this:

[image]

The above has "SEN" and power wattage and temperature as two separate menu bar entries. It would be great if we have the option of merging the two :)!

It would be also great if there was a way to re-arrange the sensors that appear on the menubar (e.g. in my case, I want to move the temperature to the top and wattage to the bottom).

**Details:**
 - Device: Macbook Pro 14 M1 Pro
 - macOS: 12.2.1
 - Application version: 2.7.6

exelban, 2022-03-13:
Hi. Thanks for the feedback.
I'm not sure it's needed. Other widgets have this option because the same widget could be enabled for multiply modules. It allows to identificate the widget. And the widget in the Sensor module could be enabled only in this widget. And it cannot be mistaken with another module.

exelban, 2022-03-20:
Hi. About the rearrange the sensors. It will be added someday. But I cannot give any timelines for that. It's not on the priority list.

closed 2022-03-20 by exelban