exelban/stats #874: Update dialog needs app name, icon, etc

ronin712, 2022-03-13:
When the dialog pops up to tell you there's a new version, there's no indication what app it's for.  It needs the app name and preferably the icon.

exelban, 2022-03-13:
Do you talk about notification alert?

rswail, 2022-03-18:
This sounds like the same issue I raised yesterday. When the update dialog pops up, it doesn't identify what the update is for. With lots of different menu bar and dock apps running, these dialogs can show up at random times, so having it say "An update for Stats is available" would be enough to at least know which app is asking :)

exelban, 2022-03-18:
@rswail yes, it's the same ticket (I suppose). But still, I'm not sure do you talk about notification alert or popup window?

rswail, 2022-03-20:
Not sure, I remember that I closed other apps and windows and there was 
a "A new version is available Install/Cancel?" dialog.

Ron

rswail, 2022-03-22:
[image]
This is the dialog that I found "lurking" behind my other windows/apps when I minimized them. I know now that it's for Stats, but there's nothing to indicate it.

rswail, 2022-03-22:
[image]
Followed by this one after clicking Download. Both need some sort of indication that it's for the Stats application :)

closed 2022-04-03 by exelban