exelban/stats #886: Connects to external web site when Network module remains disabled

macbugs, 2022-03-19:
[image]

exelban, 2022-03-20:
Hi. Very informative description.
But still, I have rechecked if there is a request to obtain an external IP when the network module is disabled. And no, there is no request if the network module is disabled.

macbugs, 2022-03-20:
Currently I only have these modules enabled: GPU, RAM, DISK, SENSORS
But still I see recurring firewall alerts many times each day.
It also wants to connect to `api64.ipify.org`

exelban, 2022-03-20:
Ohh. Ok, it's a public IP auto update mechanism. Will fix that.

macbugs, 2022-03-21:
ok, I understand now... I had it set to check for updates once per week, but it actually happens once per hour

closed 2022-03-27 by exelban