exelban/stats #889: updating to 2.7.8 is not working properly

movi62, 2022-03-22:
running 2.7.7, downloading 2.7.8 (via update-dialog in preferences), installing. afterwards 2.7.7 is still running and informing me about a new update being available...

system: MacBook Air M1 (2020), 8core CPU+GPU, Montery 12.3

exelban, 2022-03-22:
Hi.
Do you have the `Stats.dmg` file in the `Downloads` folder?

movi62, 2022-03-22:
the process is as follows:

1. preferences - check for updates
2. new version available "current version v2.7.7" - "newest version v2.7.8"
3. download (afterwards the stats.dmg file is in the download folder)
4. install (by using the button in the dialog)
5. stats is terminated and starts again (in the old version)
6. message announces new version is available
7. stats.dmg disappeared from the download folder

exelban, 2022-03-23:
Thanks. It's very strange. Could you maybe post the downloaded `Stats.dmg`? (run the update process and make a copy meanwhile)

ISKOTB, 2022-03-23:
> running 2.7.7, downloading 2.7.8 (via update-dialog in preferences), installing. afterwards 2.7.7 is still running and informing me about a new update being available...
> 
> system: MacBook Air M1 (2020), 8core CPU+GPU, Montery 12.3

same here and I never had stats.dmg in the download folder!? 
Thank You

exelban, 2022-03-27:
Hi. Sorry, but I cannot reproduce this bug on any of my macs. Maybe it's related to your permissions or something else. 
@ISKOTB it means that the new version could not be downloaded.

closed 2022-03-27 by exelban

movi62, 2022-03-29:
update is working now fine. (2.7.9. --> 2.7.10). thanks!!
tested on Monterey and Catalina