exelban/stats #892: Missing average and hottest CPU sensors

rajtslegr, 2022-03-28:
In 2.7.9 average and hottest CPU are missing from the list.

[image]
[image]

**Details:**
 - Device: MacBook Air M1
 - macOS: 12.3
 - Application version: 2.7.9

TangYuanqing, 2022-03-28:
I meet same problems at same environments. Meanwhile, I couldn't switch type of modules any more, when i click another type of module, it have no response.

exelban, 2022-03-28:
Hi. Thanks for the feedback.
Yeap, there is a bug, will be fixed.

rajtslegr, 2022-04-03:
I don't think this is fixed in v2.7.11.

mikemike396, 2022-04-03:
> I don't think this is fixed in v2.7.11.

Not seeing it in 2.7.11 either. 

[image]

mervenca, 2022-04-03:
Yep, still missing those aswell..
Kind of offtopic, but which of those could be the most accurate for cpu temp monitoring, until its going to be fixed?

exelban, 2022-04-04:
Hi. Could someone check this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/8412503/Stats.dmg.zip)

rajtslegr, 2022-04-04:
Looks good!

[image]

rakleed, 2022-04-04:
@exelban Also, on a Mac with a M1, the `Average SOC` and `Hottest SOC` settings are not updated. That is, they read the value when the app starts, and do not change it anymore.

In version 2.7.8 everything worked correctly. The version from [this comment](https://github.com/exelban/stats/issues/892#issuecomment-1087916216) also doesn't solve this problem.

mervenca, 2022-04-04:
Yep, works here on M1 air aswell, thanks!!

exelban, 2022-04-05:
@rakleed ok, I will take a look at this.

closed 2022-04-10 by exelban