exelban/stats #899: Stats.app doesn't run after macOS Upgrade

smithjw, 2022-04-02:
I'm not sure if anyone else has seen this, but the last two times, I've upgraded macOS (to 12.3 and to 12.3.1), the Stats app will no longer launch and I'll need to reinstall it. 

Looking into the Applications folder I'm seeing the Stats app appearing as a broken app 
[image]
Digging deeper though, looking within that app bundle, I see the application still there
[image]
Though [Apparency](https://mothersruin.com/software/Apparency/) doesn't show this as a correctly signed app
[image]
If I drag child Stats.app outside of the broken one in the Applications folder, it shows as an invalid app bundle
[image]
The child Stats.app bundle will launch but given it doesn't show as a valid app, it's not best to simply move it outside of the broken Stats.app in the Applications folder and instead reinstall from the downloaded DMG.

Looking at the app within the DMG downloaded from the Releases page, I'm seeing the app bundle correctly signed and verified 
[image]

exelban, 2022-04-03:
Hi. Thanks for the feedback.
I'm installing macOS 12.3.1 right now and will check this. But it looks like some bug in macOS. I had this type of problem a few times with different apps after the update.

exelban, 2022-04-03:
I have updated my mac from 12.3 to 12.3.1. And didn't notice any problem with Stats after the update. So it's not Stats related bug, but macOS.

closed 2022-04-03 by exelban