exelban/stats #900: CPU/GPU widgets not showing on macOS 12.3 M1

wyan, 2022-04-03:
**Describe the bug**
The CPU / GPU / Temperature widgets are not showing on the menu bar after updating to macOS 12.3, only the battery widget. Turning them on and off, or reinstalling Stats don't fix the issue.

**Details:**
 - Device: MacBook Air M1
 - macOS: 12.3
 - Application version: happening on 2.7.11 and a few previous updates

exelban, 2022-04-03:
Hi. Thanks for the feedback.
Did you have enabled these modules? If yes, what widgets you have enabled in each module?

wyan, 2022-04-04:
I had CPU and GPU temperature, which now don't exist anymore. I can choose a different sensor and the temperature displays fine. But the CPU and GPU usage graphs don't show anymore.

exelban, 2022-04-04:
Ok, first of all. Did you have enabled HID sensors option?

ryneandal, 2022-04-04:
I can second this, and I have HID sensors and the Sensors module enabled:

[image]
[image]

Also, it is the only module where the widgets tab is empty:

[image]

Let me know how else I can help.

ryneandal, 2022-04-04:
Nevermind, I looked at code and it appears to be user error on my end, I didn't realize I had to drag the widgets beyond that divider to enable. Ignore me, apologies :)

exelban, 2022-04-05:
You must have enabled at least one widget. No problem)

wyan, 2022-04-05:
Oh, I hadn't noticed this change in behaviour either. Dragging a widget to the other side works. Sorry about that! :)