exelban/stats #909: You cannot open the application "Stats" because it may be damaged or corrupted

1Mark, 2022-04-12:
**Describe the bug**
A clear and concise description of what the bug is.
Each time I reboot I get
You cannot open the application "Stats" because it may be damaged or corrupted
[image]

Reinstalling fixes it but this is the second time in 2 days.

```
error	13:38:14.286211+0100	Dock	LAUNCH:Returning kLSNoExecutableErr because bundle has no inode, (null) node=<private> status=-10827
error	13:38:14.286699+0100	CoreServicesUIAgent	handle LS launch error: {\n    Action = oapp;\n    AppPath = "/Applications/Stats.app";\n    ErrorCode = "-10827";\n}
```

**Details:**
 - Device: MacBook Pro (13-inch, 2020, Four Thunderbolt 3 ports)
 - macOS: Big Sur
 - Application version: 2.7.11

1Mark, 2022-04-12:
> Hi. Thanks for the feedback.
> Could you please share the damaged copy of the Stats?

How can I share it?

1Mark, 2022-04-13:
[Stats.zip](https://github.com/exelban/stats/files/8479379/Stats.zip)

exelban, 2022-04-13:
Thanks for sharing. It's not a bug in Stats. Somehow the app is inside the app. You can get the app by: right click to the file you shared -> Show Package Contents. Inside you will find the app.

closed 2022-04-13 by exelban

1Mark, 2022-04-13:
@exelban it's very weird. Managed to recover it without reinstalling by what you said but it's very weird. Not even sure what to Google so I can understand how and why it happens.

sametsafak, 2022-04-26:
I also have this problem. I am using 2 accounts with my mac, maybe it's related. Second account is not an admin account. Maybe this is the reason.

exelban, 2022-04-26:
@sametsafak thanks for the feedback. I will create a few accounts and will check that.

1Mark, 2022-04-26:
I only have one account

exelban, 2022-04-26:
So the problem is not in the multiple accounts(

1Mark, 2022-04-27:
the issue happened again :(