exelban/stats #910: Trailing '0' removed from kernel_task memory

morphadl, 2022-04-14:
When kernel_task memory ends in a '0' that '0' is not displayed, e.g. 330MB appears as 33MB or 400MB as 4MB. Any other final digits seem to be displayed correctly as do other processes.

kernel_task is still shown in its correct position in the list, right now I have it showing between WindowSerever (455MB) and FirefoxCP... (215MB).

Stats v2.7.12, macOS v12.3.1

exelban, 2022-04-14:
Hi. Thanks for your feedback.
Could you please share the output of `top -l 1 -o mem -n 8 -stats pid,command,mem`?

morphadl, 2022-04-14:
```
user@host ~ % top -l 1 -o mem -n 8 -stats pid,command,mem
Processes: 510 total, 2 running, 508 sleeping, 2221 threads 
2022/04/14 18:12:20
Load Avg: 1.58, 1.47, 1.43 
CPU usage: 3.73% user, 11.68% sys, 84.57% idle 
SharedLibs: 582M resident, 87M data, 212M linkedit.
MemRegions: 85531 total, 2641M resident, 335M private, 1959M shared.
PhysMem: 25G used (2825M wired), 7016M unused.
VM: 17T vsize, 3135M framework vsize, 0(0) swapins, 0(0) swapouts.
Networks: packets: 761119/857M in, 417441/62M out.
Disks: 853260/14G read, 536589/24G written.

PID    COMMAND          MEM  
1521   firefox          1119M
166    WindowServer     503M 
12105  plugin-container 318M 
0      kernel_task      270M 
8284   plugin-container 159M 
7797   plugin-container 136M 
98     RTProtectionDaem 135M 
12265  plugin-container 126M 
```

morphadl, 2022-04-16:
I beleive the error is this line in stats/Modules/RAM/readers.swift...
152                var command = str.replacingOccurrences(of: pidString, with: "")

The kernel_tasks process ID is 0, this is being stripped from the memory reading as well as removing the pid at the start of the line. I suspect the same would happen for any process where its RAM usage contains the process ID, e.g. pid 98 with RAM of 398MB or process ID 576 using 576MB.

closed 2022-04-17 by exelban