exelban/stats #913: [Feature Request] Notification on high CPU, low RAM, etc.

daitarn, 2022-04-19:
Howdy, first let me thank you for this great app!

So a notification window alerting of high CPU usage by a single app, low RAM, disk etc., by manually setting a threshold:
- CPU usage by a single app over 50% of CPU for 20 seconds
- RAM bellow 20%
- disk space below 20%

CPU by a single app is what I'm looking for, but I thought that could be a useful feature for everything else.

Cheerios!

exelban, 2022-04-24:
Hi. Thanks for the feedback.
I think it will be a good feature. Will be added.

closed 2022-07-10 by exelban

daitarn, 2022-07-11:
Thanks for implementing this feature, but it's missing the part where we can set a timeframe in which the high usage occurs.

Meaning, CPU over 50% for at least 10 seconds.

The the CPU constantly jumps at high rates for brief times. I meant it to letting me know if an app uses high CPU for a prolonged period of time.

Thanks!

exelban, 2022-07-11:
Hi. It will be added in the future. But I'm cannot say when exactly.