exelban/stats #916: [MB Pro 2019] High memory consumption

elohhim, 2022-04-22:
**Describe the bug**
Stats build up 1,2GB memory consumption on my system.
[image]

Not sure when it happened, could be gradually over time as I have ~24days uptime.

Modules enabled:
- RAM
- BATTERY
- DISK
- SENSORS

I was on Stats 2.7.9, I am testing 2.7.13 now but it might take some time if this is related to long uptime. Anyways from the change log I can't see anything that could fix potential memory leak between those versions.

**Details:**
 - Device: MacBook Pro 16" (Late 2019)
 - macOS: 11.6.0
 - Application version: 2.7.9

exelban, 2022-04-22:
Hi. Thanks for your feedback.
When the next time you will notice such high memory consumption. Please disable one by one modules and check if the consumption will go down. It will help to detect what exactly module has a memory leak.

elohhim, 2022-04-22:
Sure thing :) I will monitor memory usage for few days and see if it starts building up again, then try to pinpoint module.

elohhim, 2022-04-26:
Okay so it has been 4 days of uptime and Stats build up to ~300MB memory usage: 

[image]

I went through modules one by one and disabling none of them shows any significant instant drop in memory consumption. 

I've disabled SENSORS for now (as I can see that it was already cause of memory issues in the past) and will check if consumption continues to rise.

lowercase00, 2022-05-02:
I have noticed a huge increase in CPU consumption in the last few days, to the point the computer is very slow.
I will try to understand a bit better which module may be causing this, but also noticed I'm on 2.7.0 (it seems the the check for update is not working very well I guess?) will try updating as well.

[image]

elohhim, 2022-05-05:
Still leaks with:
- RAM
- BATTERY
- DISK

[image]

exelban, 2022-05-05:
strange, because I have running Stats for a few days with all modules. And usage all time around 80MB.

exelban, 2022-05-24:
Hi.
I have tested the app for a few weeks. And didn't notice any memory increase. Also, there are no memory leaks or a big increase in the monitoring tool. I have tested that on apple silicon as on intel based.
If you want to help you can run the Stats in the Xcode and debug what exactly causes the leak in your case. 

I thought that the leak could be in the Sensors module, but @elohhim confirms that there is a leak even without it.

If you have more details or information about what caused the leak, give me know. I will reopen the issue.

closed 2022-05-24 by exelban

exelban, 2022-05-25:
Thanks, I will take a look at the Disk module. Will give feedback if find something.