exelban/stats #918: [Feature Request] Exclude "process base" Network system to 2 or 3 items

mhkanon, 2022-04-26:
Hi stats team, First of thanks to stats team to give us a nice app. I use it for 1 year+. I want a new feature in the "**network**" section. On my laptop when I select "**Reader type: Interface Base**" It only shows "**Upload speed**". That's why I use "**Reader type: Process Base**. But in "**Reader type: Process Base** I face a problem speed double show. So, I need to **exclude** **one or two** **process** speeds. If I exclude that process I got my real speed. See the screenshot you got it clearly. Thanks
[image]
[image]

mhkanon, 2022-04-29:
**4 days ago** I posted. But no one replied. I saw a new post after I posted it replyed. @exelban

exelban, 2022-04-29:
Hi. Yes, sorry for the no response. I'm thinking about it.
From one side I don't see any reason to exclude some processes from the calculation. But I agree that it could be helpful. But again I don't see a good way of how it could be realized from ui/ux part.

mhkanon, 2022-04-29:
It is **helpful** for me so much who use **ssh**. See If I **exclude ssh** I got the **real speed**. Please, add this option. Thank you.

mhkanon, 2022-04-29:
> 

It is **helpful** for me **so much** who use **ssh**. See If I **exclude ssh** I got the **real speed**. Please, add this option. Thank you.

exelban, 2022-05-03:
Hi. Sorry, but I will not add an option to remove some processes from the list. It requires too many changes. And I will not have time for that.

closed 2022-05-03 by exelban