exelban/stats #922: Main screen only option?

chris-short, 2022-04-27:
I use a two monitor setup; one is external, the other is the much smaller laptop display. It'd be really cool to add a feature for Main Screen only on the menu bar because things are super cramped on that laptop monitor.

I'm not even sure if Apple has APIs for that. It's a thought, I figured I'd bounce it off y'all.

chris-short, 2022-04-28:
The screen with the menu bar at the top as seen in System Preferences -> Displays (image attached)

[image]

exelban, 2022-04-28:
Ok, but what feature are you asking for?

chris-short, 2022-04-28:
Sorry for making this feel like teeth pulling.

Stats to show up ONLY on the main screen where the menu bar is located in the previously mentioned image.

exelban, 2022-04-28:
Yes, the menu bar is visible only on the main screen. It's how macOS works.

chris-short, 2022-04-28:
The menu bar is extended across displays last I checked (looking at it).

chris-short, 2022-04-28:
Ah, the correct term in Apple parlance is Main display. I'd like Stats to limit itself to Main display's menu bar ONLY.

closed 2022-05-01 by exelban