exelban/stats #923: Late 2013 Macbook Pro 15" (MacbookPro11,3) reported as Mid 2014

drawz, 2022-04-29:
Really just a cosmetic thing, but the late 2013 Macbook Pro 15"  (MacbookPro11,3) is reported as Mid 2014 in stats. CPU, memory, SSD, etc. all show up correctly.

exelban, 2022-05-01:
Hi. As I say there are 2 models with the same id. And I cannot detect the production year for this model correctly.

closed 2022-05-01 by exelban

drawz, 2022-05-03:
It would be kind of an annoying workaround, but you could look at the CPU used as this is the primary difference between the two years. There's a nice table in this EveryMac article: https://everymac.com/systems/apple/macbook_pro/macbook-pro-retina-display-faq/differences-between-macbook-pro-retina-display-late-2013-mid-2014.html

(The stock SSDs might be different also, but those are removable, so not a good choice to differentiate)