exelban/stats #926: Sensors widget flashing white background

briankosw, 2022-05-02:
**Describe the bug**
The sensors widget keeps flashing white. I've tried switching modes and changing the widget's settings, but I still the flashes. This doesn't happen for the other widgets.

[image]

**Details:**
 - Device: MacBook Pro 2015
 - macOS: macOS Catalina
 - Application version: 2.7.16

exelban, 2022-05-05:
Thanks, thanks for your feedback.
I already saw this type of bug. Unfortunately, I cannot fix it because I don't know what triggered this flashing.

briankosw, 2022-05-06:
I see. For what it's worth, I only see this behavior for the sensors widget. Hopefully that helps you narrow down the root cause.

closed 2022-05-08 by exelban