exelban/stats #928: Customizable sensor order (at least rudimentary)

porg, 2022-05-03:
I'm aware that the order of the different menu item sections is controlled by macOS.

Where I am proposing an improvement is the order of the different sensor values chosen for display.

1) Minimalistic improvement: Order of adding items makes a difference (currently order is hardcoded)

- a) Sensor items get added and removed in the order that the items get toggled.

- b) Additional options which control how the sorted list items get positioned:
Layout: One Row | Two Rows
Growth: Right to left | Left to right
Wrapping: Fill row then wrap | Fill column then wrap (Zig-Zag) 

2) Ideal implementation: WYSIWYG

Layout : ( ) One row   (•) Two rows

#### Drop Items here
Row 1:  __________  __________  __________  __________  __________  
Row 2: __________  __________  __________  __________  __________ 

#### Items
Sorted in topical sections as draggable token-items (with their variable name)

exelban, 2022-05-03:
Hi. Thanks for your feedback.
Yes, I have a plan to add order to the sensors module.

closed 2022-09-11 by exelban

porg, 2024-10-24:
# QA Test for sensor widgets
Test result for version 2.11.15

Precondition: Be sure the global "Sensors" is activated

## I) Set the sensor order in implicitly by order of activation
1. Go to panel "Sensors" →  tab "Module"
2. Activate the sensors in the order you want them to appear (if on 2 lines, then in a zig zag top down, left to right).
3. Quit the app and restart it.
    - ❌ Order is not persisted!
    - Feature improvement, see: link-to-issue-later

## II) Set the sensor order explicitly
1. In panel "Sensors", switch to tab "Widgets"
2. Activate the sensors you want in the widget.
3. Switch to tab "Widgets".
4. Drag the sensors into your desired order.
    - Has multiple usability issues, see: link-to-issue-later
5. Quit the app and restart it.
    - ✅ Order order as when quitting the app.
7. Settings: Export, reset, import.
    - ✅ Sensor order as it has been at export time.