exelban/stats #929: Add Clock option

sfc-gh-afedorov, 2022-05-03:
For our Ops friends and hobbyists, having UTC and local timestamps on two lines would be ideal. The way iStat Menus does it [1] seems all right, but something simple like the Wifi Signal details pane can work just as well:

[image]

1. https://bjango.com/help/istatmenus6/time/

exelban, 2022-05-05:
Hi. Thanks for the feedback.
I already see a proposal for a clock option. But your idea with 2 rows configuration looks nice. I suppose it could be added to the app. But it not will be in the near future. I will create a new module for that.

usergitty, 2022-06-30:
I already like this app more than iStats, it's just missing the time module. It's my favorite module in iStats. Really handy to have world clocks and calendar a click away. And the menubar icon is customizable to show whichever timezone you want.

closed 2023-04-02 by exelban

sfc-gh-afedorov, 2023-04-04:
Super cool, thank you! Added a few post-commit comments, but really appreciate the attention to this feature. iStats Menus is uninstalled :)