exelban/stats #933: RAM widget and popup don't show the same value

valynor, 2022-05-08:
The RAM % widget and the value shown in the popup are sometimes different (off by 1%, rounding error?). Update interval is 1 second, it's not that. 

[image]

exelban, 2022-05-09:
Hi, Thanks for your feedback.
I will recheck this. It could be or different rounding mechanism. Or update mechanism (the popup is not updating when closed).

zigapovhe, 2022-05-17:
Hi! I can replicate same thing with CPU widget. My guess would also be different rounding mechanism because both values are updating at the same time (it just differs in value sometimes). Network widget looks fine.

[image]

closed 2022-05-29 by exelban