exelban/stats #936: [Feature Request] Colorize network "Speed" value text to match pictogram colors

fiveangle, 2022-05-12:
In order to more elegantly and compactly visually indicate network activity, is it possible to add a "Colorize Speed Values" (or whatever you feel is best to describe it) option under the Network Widget settings for when Speed widget is selected ?  This will allow those that desire a smaller menubar footprint to select "None" for Pictogram, still get a colorized visual indicator of bandwidth direction as the colorized Speed values they change.

To be clear, the idea is to allow an option to colorize the white text values below to match the Pictogram indicator colors, even when "None" is selected in the Pictogram settings:

[image]

This is not a new idea of mine, but a feature that I appreciated with the now-defunct MenuMeters by Alex Harper:

[image]

Thanks for such a more full-featured and lovely utility that I doubt I would have ever stumbled across if not for MenuMeters becoming obsoleted !

exelban, 2022-05-22:
Hi. Thanks for your feedback.
I think it could be added.

closed 2022-06-05 by exelban

fiveangle, 2022-06-12:
It's glorious. Thank you ! ❤️