exelban/stats #942: repeated crashing on a multi-user mba with corp management profiles

taizoc, 2022-05-18:
Stats Version 2.7.17
MacBookAir7,1 - i5 8GB 
Two accounts on this device. On my account with management policies, Stats repeatedly quits when I try to pull up Stats settings or modify menu item location. 
When stats comes back, widgets will be duplicated. I'll have 2 CPU visualizers, or RAM, etc.

exelban, 2022-05-23:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in `Applications -> Utilities -> Console`. Find the Crash Reports tab. There you must see a row called Stats.

closed 2022-05-29 by exelban

sndrs, 2022-05-30:
I have something very similar. not managed, but two accounts, both run stats.

one account (I'm guessing one without sufficient permissions?) gets stuck in an update loop:

https://user-images.githubusercontent.com/867233/171038783-732399f2-bc69-4b81-b5d1-94a90c167350.mov

https://user-images.githubusercontent.com/867233/171038903-b003067a-2f74-479c-b235-34fcf081857b.mov

it doesn't appear to crash as such, just fail to update and gets stuck in a loop

sndrs, 2022-05-30:
there's no way out of this loop, but when you log out, the update window flashes up, asking for a password for the update. the machine logs out before you can do anything. and once you log back in, you find a new crash report:

```shell
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [37905]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.7.14 (359)
Code Type:             X86-64 (Native)
Parent Process:        Exited process [37902]
User ID:               501

Date/Time:             2022-05-30 18:35:39.3244 +0100
OS Version:            macOS 12.4 (21F79)
Report Version:        12
Anonymous UUID:        C327AA55-8874-58BD-79DE-E326C414D644

Sleep/Wake UUID:       FE6D62A5-356E-487B-BA29-DDB4B527F824

Time Awake Since Boot: 27000 seconds
Time Since Wake:       16623 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff80d1ed00e __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff80d2231ff pthread_kill + 263
2   libsystem_c.dylib             	    0x7ff80d16ed24 abort + 123
3   HIServices                    	    0x7ff812d5f7d8 _RegisterApplication + 13413
4   HIServices                    	    0x7ff812d5c286 GetCurrentProcess + 23
5   HIToolbox                     	    0x7ff815f62a0f MenuBarInstance::GetAggregateUIMode(unsigned int*, unsigned int*) + 63
6   HIToolbox                     	    0x7ff815f62997 MenuBarInstance::IsVisible() + 41
7   AppKit                        	    0x7ff80fcefcbf _NSInitializeAppContext + 35
8   AppKit                        	    0x7ff80fceda12 -[NSApplication init] + 417
9   AppKit                        	    0x7ff80fced655 +[NSApplication sharedApplication] + 120
10  Stats                         	       0x10e33ea85 0x10e339000 + 23173
11  dyld                          	       0x11291851e start + 462

Thread 1:
0   libsystem_pthread.dylib       	    0x7ff80d21ef48 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	    0x7ff80d21ef48 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000112993600  rcx: 0x00007ff7b1bc5408  rdx: 0x0000000000000000
  rdi: 0x0000000000000103  rsi: 0x0000000000000006  rbp: 0x00007ff7b1bc5430  rsp: 0x00007ff7b1bc5408
   r8: 0x0000000000000000   r9: 0x0000000000000090  r10: 0x0000000000000000  r11: 0x0000000000000246
  r12: 0x0000000000000103  r13: 0x0000000000000000  r14: 0x0000000000000006  r15: 0x0000000000000016
  rip: 0x00007ff80d1ed00e  rfl: 0x0000000000000246  cr2: 0x0000600000cec000
  
Logical CPU:     0
Error Code:      0x02000148 
Trap Number:     133

Binary Images:
    0x7ff80d1e5000 -     0x7ff80d21cfff libsystem_kernel.dylib (*) <61711d11-e776-3bc3-b9a2-6f9f37cb8499> /usr/lib/system/libsystem_kernel.dylib
    0x7ff80d21d000 -     0x7ff80d228fff libsystem_pthread.dylib (*) <bc574849-1aae-31e7-b350-916dda999d97> /usr/lib/system/libsystem_pthread.dylib
[... 171 lines cut ...]
  },
  {
    "source" : "P",
    "arch" : "x86_64",
    "base" : 4606472192,
    "size" : 442368,
    "uuid" : "b70ce1ec-b902-3852-8268-05de00bfa8d5",
    "path" : "\/usr\/lib\/dyld",
    "name" : "dyld"
  }
],
  "sharedCache" : {
  "base" : 140703345680384,
  "size" : 15220686848,
  "uuid" : "398acfb4-57f6-31e0-bc82-e9959e5c92ce"
},
  "vmSummary" : "ReadOnly portion of Libraries: Total=969.8M resident=0K(0%) swapped_out_or_unallocated=969.8M(100%)\nWritable regions: Total=750.5M written=0K(0%) resident=0K(0%) swapped_out=0K(0%) unallocated=750.5M(100%)\n\n                                VIRTUAL   REGION \nREGION TYPE                        SIZE    COUNT (non-coalesced) \n===========                     =======  ======= \nColorSync                            4K        1 \nKernel Alloc Once                    8K        1 \nMALLOC                           237.2M       34 \nMALLOC guard page                   24K        5 \nMALLOC_MEDIUM (reserved)         120.0M        1         reserved VM address space (unallocated)\nMALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)\nMemory Tag 241                     296K        4 \nObjC additional data                15K        1 \nSTACK GUARD                       56.0M        3 \nStack                             9232K        3 \nVM_ALLOCATE                          8K        2 \n__CTF                               756        1 \n__DATA                            21.6M      384 \n__DATA_CONST                      19.7M      233 \n__DATA_DIRTY                      1137K      145 \n__FONT_DATA                          4K        1 \n__LINKEDIT                       648.7M       17 \n__TEXT                           321.2M      398 \n__UNICODE                          592K        1 \ndyld private memory               1024K        1 \nmapped file                       29.2M        2 \nshared memory                       24K        2 \n===========                     =======  ======= \nTOTAL                              1.8G     1241 \nTOTAL, minus reserved VM space     1.3G     1241 \n",
  "legacyInfo" : {
  "threadTriggered" : {
    "queue" : "com.apple.main-thread"
  }
}
}

```

exelban, 2022-05-30:
@sndrs thanks for the feedback. Update of Stats does not require any special permission. It need only access to the `Download` and `Applications` folders. But it does not need a root for sure. I will try to recreate this behavior. Maybe you have special permissions for each account?

sndrs, 2022-05-31:
They're both admin accounts, but nothing else special about them I don't think

taizoc, 2022-06-01:
Hi there. Apologies, I opened this ticket and then left for a week of vacation - sorry for ghosting the thread. 

https://user-images.githubusercontent.com/26072224/171335434-18ca0d8d-6b47-4d47-b3c4-4e67c5564794.mov
I don't think there's any important PII in that screen recording... 

Please let me know if there is another place I can find relevant logs!