exelban/stats #948: a closed process keep showing in high occupancy process list

faimin, 2022-05-29:
[image]

Stats Version: 2.7.17
OS: Mac OS 12.4 (Intel)

exelban, 2022-05-29:
Hi. Thanks for the feedback.
It only means that `top` still returns that process. Or you have a long top process update interval and it is just not updated yet.

closed 2022-05-29 by exelban

faimin, 2022-05-29:
Do you mean this is a system API problem?  It does not showing in `Activity Monitor.app`, and the Electron process was still showing in `Stats` 10 hours later, other processes update normally.

exelban, 2022-05-29:
I will take a look to see if it could be an internal bug