exelban/stats #950: CPU/GPU temperatures always in Celsius on 2.7.18

thisiszquake, 2022-05-29:
**Describe the bug**
CPU/GPU temperatures show in Celsius on 2.7.18, despite having the preferred unit set to Fahrenheit. If the preferred unit is changed to Celsius, it tries to convert the Celsius temperatures (which it thinks are in Fahrenheit) into Celsius, resulting in even lower temperatures being displayed.

2.7.17 (intended):
[image]

2.7.18:
[image]

The screenshots were taken within one minute of each other.

Downgrade to 2.7.17 for a temporary fix.

**Details:**
 - Device: MacBook Pro (13-inch, 2020, Four Thunderbolt 3 ports, Core i7)
 - macOS: 12.4
 - Application version: 2.7.18

exelban, 2022-05-29:
Hi.
Thanks for the feedback. The measurement units for temperature work fine if you switch between Celsius and Fahrenheit.
Could you please explain what exactly does not work correctly?

arandorion, 2022-05-30:
I have always had the temperature unit set to F. However, after this latest update, the temperature is displayed with the C value and the F label. I can find no way to switch between C and F. See attached image.

Thanks

[image]

arandorion, 2022-05-30:
I found where to switch between C and F, but it has no effect.

Thanks

sparksparker, 2022-05-30:
it looks like the current fahrenheit temperature is actually in celsius and needs to be calculated to be in fahrenheit again

exelban, 2022-05-30:
Ohh, I think I know the problem. Will try to fix that.

exelban, 2022-05-30:
Could please someone confirm that this build works fine?

[Stats.dmg.zip](https://github.com/exelban/stats/files/8797265/Stats.dmg.zip)

thisiszquake, 2022-05-30:
Yes, this build works fine now.

exelban, 2022-05-30:
Thanks for the feedback.
I have released a [hotfix](https://github.com/exelban/stats/releases/tag/v2.7.19) with that fix.

closed 2022-05-30 by exelban

arandorion, 2022-05-30:
Thanks for the quick fix!