exelban/stats #954: Stats 2.7.19 on MacMini6,1 small UI glitch due to many sensors (incl. CPU core doubles)

mackonsti, 2022-06-04:
**Describe the bug**
On my MacMini6,1 running latest Monterey, the latest Stats 2.7.19 has a small UI problem in the Sensors tab:
a) The hidden vertical scroll bar is on top of elements (wasn't perhaps considered?)
b) The CPU1 and CPU2 etc. entries are seen as CPU Core X and CPU Core X twice.

[image]

**Details:**
 - Device: MacMini6,1
 - macOS: Monterey 12.4
 - Application version: 2.7.19

**UPDATE:** This scroll-bar glitch also happens on Catalina on MacMini8,1... perhaps consider moving the switches and drop-down list further inside the window?

[image]

**Notes:**
Thank you for your great work on this tool, I'd like to express my appreciation.

In order to help you, please provide a Terminal (CLI) command to list for you the total sensors of this Mac model, so that you can understand why some CPU sensors show twice under the same name. On both MacMinis I get twice the same "CPU x Core"

Also, in order to keep consistency with labels, I'd suggest you capitalise the text labels of the first sensors as the later sensors appear capitalised, anyway.

Thank you.

exelban, 2022-06-04:
Hi. Thanks for the feedback.
Yeap, on some machines there are doubled CPU cores. It's because macOS reports two sensors: `TC%c` and `TC%C`.
I cannot remove some of them, because some mac has only one of them. And I'm not sure if there are the same sensors if both are presented.

The glitch with the scrollbar. Yeap, I know about it. It's because there are two different views for module settings and sensors. But I'm not sure worried about that. I really do not have time to fix it( Maybe someday a PR with that fix will appear. But I'm pretty sure I will not do that.

Now about the CLI tool. There is one.
Get all sensors: `/Applications/Stats.app/Contents/Resources/smc list`
Get fans: `/Applications/Stats.app/Contents/Resources/smc fans` 
I suppose that all commands you can find with `help` command.

closed 2022-06-04 by exelban

mackonsti, 2022-06-05:
Thank you for your time and response @exelban 

I understand time constraints for such a non-functional i.e. UI glitch that's expected, it seems, from you.
Hopefully you can continue work on Stats in the coming months!

And yes I understand about the sensors, it's better to keep them all listed rather than excluding some as you said, especially since `TC%c` and `TC%C` are not really different...?

A last comment that I had forgotten to ask-request.

Would it be possible to **add the units for the fan sensor i.e. speed** possibly via a switch on/off? I mean the `rpm` word. It would be useful to have it displayed as a plain fan number is not too useful, if you agree.

Can I post a new ticket here on GitHub for this?
Thank you again.

exelban, 2022-06-05:
Hhh, yeah the trick with capital C could be nice. I will add that.

Nope, it's not possible. It will not fit in the widget. Otherwise, the widget will be too wide when the rpm will be added. There was an issue with a description of why the width could not be dynamic etc.