exelban/stats #957: Stats Application Weird Issue

rohitcoder, 2022-06-06:
**Describe the bug**

Stats is having some issues, this never happened before and this is happening with only this "Stats application". It appears and disappears and there is no way to kill this process, or tell me way to uninstall this, or to remove from startup.

BTW, I updated my MaC few days ago, this could be the main reason?

**Screen Recordings:**

https://user-images.githubusercontent.com/17665703/172101626-59fac8dd-f2e0-476e-ad87-7be12537f67d.mov

https://user-images.githubusercontent.com/17665703/172101555-dfa38e4e-1643-41e0-8b9c-7f4902db6c0d.mov

 - Device: Macbook Pro M1, 2020
 - macOS: MacOs Monterey 12.4
 - Application version: Downloaded this 2 weeks ago, can't figure out version because of above mentioned issue, this disappears...

exelban, 2022-06-06:
Hi. Thanks for the feedback.
It's very strange. Looks like app goes into update loop: it detects a new version -> download it -> install -> running the new version -> and again detecting a new version. On posted film visible that the new .dmg appears, and after disappears. It means that the new version is run and cleans up the installation files.
I will add a code that will prevent this kind of loop. But it will be available only with the next update. I don't see a way how I can prevent it on your side. Maybe a local block of `https://api.serhiy.io/v1/stats/release/latest` and `https://api.github.com/repos/exelban/stats/releases/latest` will help. Because the app will not receive info that a new version exists.

I have no idea how it is possible. Could you please post the `tag_name` from the `https://api.serhiy.io/v1/stats/release/latest`?

rohitcoder, 2022-06-07:
Hi @exelban 

Thanks for the reply, just noticed this.

Actually, I do pen-testing and I use/build multiple tools which send excessive requests to targets, it seems my IP got flagged by Cloudflare and that's the reason I'm getting this error there. You can fix this on your side :)

[image]

closed 2022-06-07 by exelban

rohitcoderCdefense, 2022-06-07:
Hi @exelban,

I'd to comment on this thread from my office Github Account, Since you blocked my main account from further communication. I typed a whole long thread (from my phone) to explain in detail and what you got wrong here. 

After submitting, i got error and then i knew i'm blocked.

I won't be able to explain in detail now, i'll just explain in short here.

I said i am pentester and i test targets, that doesn't means i tested your application and the question is why i'll do that? Is there any point of that ? (Any monetary reward? No Any Personal gains? No) So there is no point to test your application.

You are using CloudFlare and i am also using that in my multiple products, CloudFlare shares kind of threat intelligence data with multiple account holders of CloudFlare, So it works like this.

If Site A and Site B both are using CF and i tested Site A (By sending lot of requests) then even if i Visit Site B for first time i'll get blocked by Site B, if they're using "I'm in Attack mode" or if Firewall rules are too much strict.

I am using this application since last 2 weeks, and i am really upset seeing this kind of response from an Open Source community's developer. Your profile looks impressive, but i guess you should've asked this first.

One more scenario.

This may also happen with normal users (No pentesters, or engineers) If there PC is infected with Adwares or Browser Adwares (Which opens lot of random links or Advertisement links after every clicks)  In that scenario also CF will Flag that IP and if normal users uses your service they'll face issues.

I can accept the solution of not "Using it" as you mentioned "Stop using this". You can also block this account, BTW i won't be here again, i came here just to clear everything on your doubts, because you blocked me + Removed labels from this report

Thanks,
@rohitcoder