exelban/stats #958: Sensor doesn't show cpu temps

VCoves, 2022-06-06:
**Describe the bug**
The sensors section doesn't show any CPU temperature on this device. However I tested on the macbook pro max and I could see the cpu temps. It is my first day with this device so maybe it's missing some tools to get this info?

**Details:**
 - Device: [Macbook Air M1 2020]
 - macOS: [12.4.0]
 - Application version: [e.g. 2.7.20]
 
[image]

closed 2022-06-06 by VCoves

Helios69, 2022-06-09:
still reproducible on latest version for Air M1

Helios69, 2022-06-09:
@VCoves  Have you managed to solve your problem? Why did you close the issue?