exelban/stats #959: Popups are not up-to-date when opened

morphadl, 2022-06-07:
When opening a popup (certainly CPU and RAM, probably others) it is shown in the state from when it was last closed. Depending on the update interval it maybe several seconds before all the data is refreshed. I have RAM updating every 10s and the graph was displaying its old state for 7s after opening the popup.

 - Device: 2019 16" MBP
 - macOS: 12.4
 - Application version: 2.7.19

exelban, 2022-06-07:
Hi. Yes, if the popup is closed, some data is not updated due to power consumption.

closed 2022-06-07 by exelban

morphadl, 2022-06-07:
I'm not suggesting the popups should continue to be updated when closed. Rather, they should be redrawn (with whatever current data is held (for graphs) or instant readings) the moment the popup is opened again.

exelban, 2022-06-07:
It's not possible. Because mostly all readers are disabled when the popup is closed. So there are no new reads when the popup is closed. Only the necessary one for the widget is running.

morphadl, 2022-06-07:
The readers for any values that is display only the current reading (e.g. load avg) should be started/kicked/whatever as soon as the popup is being opened so the popup is displayed with up-to-date values as soon as possible. I think these types of readers are working fine, at worst the display is updated pretty quickly.

The readers that have some sort of storage/graph  functionality are presumably running in the background at the user defined period (in my example earlier, 10s for the memory). Where the popup is opened just after a reading has been taken results in potentially very old data being displayed for many seconds until the next sample period. I do not see why the display cannot be updated with the sample just taken. The reader itself should not need to run again as the graph data must be being stored somewhere. We just need the plotting routines to be kicked to show the data that is already there.