exelban/stats #962: Stat menu blinking (close and re-open indefinitely)

yvele, 2022-06-09:
**Describe the bug**

Stat menu is blinking, close and re-open infinitely (even after system restart).
If I try to open Stat application (in order to change the settings), it closes itself after less than 1 second.

PS: Also how to disable Stat from launching at startup (without settings UI because the Stat app is unusable because it stops automatically).

**Details:**
 - Device: Macbook Pro (Retina, Mid 2015)
 - macOS: Monterey 12.2.1
 - Application version: The last one I guess (unable to see the version in the app because app is closing automatically)

[image]

exelban, 2022-06-09:
Hi.
Please open the `https://api.serhiy.io/v1/stats/release/latest`. What do you see in the result?

exelban, 2022-06-12:
Hi. Please update the Stats manually. The newest version of Stats contains a fix that prevents this infinite loop.

closed 2022-06-12 by exelban