exelban/stats #963: @VCoves  Have you managed to solve your problem? Why did you close the issue?

VCoves, 2022-06-09:
I'm replying to your comment below since that issue is comment restricted to collaborators. 

I fixed it by enabling the "HID sensors" on Sensors -> Module Settings. 
On a different note I think something isn't quite right, since it detects only 2 efficiency cores temps, and 7 performance cores, instead of 4 ef + 4 perf.
[image]
 
----------------------------------------------------------------------------------

@VCoves  Have you managed to solve your problem? Why did you close the issue?

_Originally posted by @Helios69 in https://github.com/exelban/stats/issues/958#issuecomment-1150963804_

VCoves, 2022-06-09:
@Helios69

exelban, 2022-06-09:
Duplicate of #944

closed 2022-06-09 by exelban