exelban/stats #967: Combined CPU?

movinimage, 2022-06-15:
Is it possible to see the CPU usage in the dropdown menu as combined rather than per-core/thread? I'm using an M1 MAX, so the percent is always listed as a portion of 1000%

exelban, 2022-06-18:
Hi.
Could you please specify the problem? Where do you see per-core/thread CPU usage, but want to see a combined?

BitesPotatoBacks, 2022-06-20:
@movinimage This is more of an issue with the way MacOS provides percentages for process usage, the system doesn't report values out of 100%.  MacOS reports maximum usages as the total of all your cores. Thus an 8-core system will have a max use of 800%, or a 10-core system (in your case, an M1 Max) will have a max of 1000%. The process usage provided by macOS is based on this maximum value. You can about it [here](https://eclecticlight.co/2022/02/24/cpu-percentage-is-misleading-on-m1-macs/) 

A possible implementation to fix this and to present data in the way @movinimage requests would be to divide each processes by the number of cores in the system. This behavior could be a triggered via a toggle...

Hopefully this is helpful!

exelban, 2022-06-24:
Hi. Sorry for the long response.
@BitesPotatoBacks have correctly noticed that it's how UNIX calculates the usage. And it cannot be obtained as an absolute value. If you are interested in this topic, please search for an issue where this behavior is described almost fully.

closed 2022-06-24 by exelban