exelban/stats #969: iStats for HD not updating

github-aaph, 2022-06-16:
Heyy, love this app, it's incredible! One thing I found not working is HD stat not updating. 

- When plugged in external drive SSD; It does show HD stat (despite I the toggle for external drive are off) and it does not update. It is stuck on whatever reading it gave me when plug it in (and it continues even thou already ejected the drive). I mean, forever, until restart. 
- The Macintosh HD also is not updating. If I should to use color "based on utilisation" it changes color but the bar when clicked still are showing the same values /free/used initially.
[image]

Thanks

exelban, 2022-06-18:
Hi.
Thanks for your feedback. I will take a look.

exelban, 2022-06-18:
@github-aaph What iStats mean?

closed 2022-06-19 by exelban