exelban/stats #976: [Feature Request] New items for sensors popup (average power, total power used)

valynor, 2022-06-20:
Hello,

I'd like to suggest to add two more items to the sensors popup in the Power section:

System average: average Watts used since the app was started 
Total Power Consumption: total kW/h used since the app was started

This would be great to better gauge the energy used by the device over time.

exelban, 2022-06-20:
Hi. Thanks for your feedback.
Yeap, I think it could be a good feature.

exelban, 2022-08-07:
Hi.
I was taking a look at that. Unfortunately, it's not possible to implement in a good way. The reason - is a power sensor. Almost every mac has a different sensor for total power usage (some do not have it at all). Also, the list of power duplicates each other (GPU 1, GPU 2, and GPU power sensor) so the power consumption could not be calculated by adding all power sensors. If I cannot be sure that I calculate the correct value, I cannot implement that. Because in most cases the calculation will be wrong.

If I have a good solution for that in the future, I will implement this feature.

closed 2022-08-07 by exelban

STRML, 2022-08-15:
I am actually a bit surprised by this closure, since Stats already supports "System Total" power by simply adding the DC In and Battery lines. Could they not simply be added up?

exelban, 2022-08-15:
`System Total` - it's just a sensor read from the SMC. It's not calculated.

Kiiyya, 2024-08-21:
Would it be possible to add a 1 minute, 5 minute, and 15 minute system power consumption average? Similar to how `htop` shows load averages for three different times as well. Btw I greatly appreciate this app :).