exelban/stats #984: [Feature Request & Bug] - Show location of current Public IP

usergitty, 2022-06-30:
Bug: The Public IP section under Network is not working for me. It just shows "Unknown" and it doesn't update when I hit refresh. I'm running macOS 11.6.7

Feature Request: Similar to iStats, ability to show the location of the currently connected IP. This is especially useful when using a VPN, it shows a flag of the country you're connected to in the menubar so you know whether you're still connected to the VPN or not at a glance.

usergitty, 2022-07-01:
It's fixed now! It was my bad, ipify was being blocked. Thanks.

Also, I see that ipify api has the option for geolocation. Any way this can be implemented?

exelban, 2022-07-01:
There was a code for that in the project. But looks like it does not work correctly, and it was disabled. Will take a look.

usergitty, 2022-07-04:
In my case, there is no country code under Public IP v4.... it just shows the IP.
And going to api.ipify.org , I also just get my IP showing and nothing else.

exelban, 2022-07-04:
So it's not possible to show the country code in your case.

closed 2022-07-04 by exelban