exelban/stats #985: [Enhancement] A hinto to show CPU frequency for Apple Silicon

TianyiShi2001, 2022-07-01:
According to the README I assume this software is currently not showing CPU frequency for Apple Silicon. However I just found a command that is able to do so (on my MacBook Pro 16' with M1 pro chip):

```bash
sudo powermetrics -s cpu_power
```

which gives the following output:

```
**** Processor usage ****

E-Cluster Power: 96 mW
E-Cluster HW active frequency: 1541 MHz
E-Cluster HW active residency:  48.17% (600 MHz:   0% 972 MHz:  37% 1332 MHz: 9.8% 1704 MHz:  14% 2064 MHz:  40%)
E-Cluster idle residency:  51.83%
E-Cluster instructions retired: 4.75774e+09
E-Cluster instructions per clock: 0.769614
CPU 0 frequency: 1618 MHz
CPU 0 idle residency:  60.59%
CPU 0 active residency:  39.41% (600 MHz:   0% 972 MHz:  10% 1332 MHz: 5.1% 1704 MHz: 6.7% 2064 MHz:  17%)
CPU 1 frequency: 1610 MHz
CPU 1 idle residency:  62.57%
CPU 1 active residency:  37.43% (600 MHz:   0% 972 MHz:  10% 1332 MHz: 5.1% 1704 MHz: 6.3% 2064 MHz:  16%)

P0-Cluster Power: 254 mW
P0-Cluster HW active frequency: 1099 MHz
P0-Cluster HW active residency:  20.92% (600 MHz:  38% 828 MHz: 4.5% 1056 MHz:  22% 1296 MHz:  17% 1524 MHz: 6.0% 1752 MHz: 3.6% 1980 MHz: 2.0% 2208 MHz: 1.4% 2448 MHz: 2.1% 2676 MHz: 1.2% 2904 MHz: .51% 3036 MHz: .52% 3132 MHz: .37% 3168 MHz: .04% 3228 MHz: .99%)
P0-Cluster idle residency:  79.08%
P0-Cluster instructions retired: 4.49883e+09
P0-Cluster instructions per clock: 1.79911
CPU 2 frequency: 1491 MHz
CPU 2 idle residency:  84.93%
CPU 2 active residency:  15.07% (600 MHz: .16% 828 MHz: .21% 1056 MHz: 5.1% 1296 MHz: 4.6% 1524 MHz: 1.3% 1752 MHz: .95% 1980 MHz: .46% 2208 MHz: .38% 2448 MHz: .68% 2676 MHz: .34% 2904 MHz: .11% 3036 MHz: .01% 3132 MHz: .07% 3168 MHz: .00% 3228 MHz: .73%)
CPU 3 frequency: 1662 MHz
CPU 3 idle residency:  91.08%
CPU 3 active residency:   8.92% (600 MHz: .04% 828 MHz: .10% 1056 MHz: 2.4% 1296 MHz: 1.9% 1524 MHz: 1.0% 1752 MHz: .93% 1980 MHz: .47% 2208 MHz: .23% 2448 MHz: .64% 2676 MHz: .47% 2904 MHz: .14% 3036 MHz: .01% 3132 MHz: .09% 3168 MHz: .00% 3228 MHz: .49%)
CPU 4 frequency: 1758 MHz
CPU 4 idle residency:  95.54%
CPU 4 active residency:   4.46% (600 MHz: .01% 828 MHz: .04% 1056 MHz: 1.1% 1296 MHz: .79% 1524 MHz: .29% 1752 MHz: .55% 1980 MHz: .31% 2208 MHz: .30% 2448 MHz: .36% 2676 MHz: .19% 2904 MHz: .06% 3036 MHz: .00% 3132 MHz: .05% 3168 MHz:   0% 3228 MHz: .36%)
CPU 5 frequency: 1819 MHz
CPU 5 idle residency:  97.17%
CPU 5 active residency:   2.83% (600 MHz: .00% 828 MHz: .02% 1056 MHz: .74% 1296 MHz: .37% 1524 MHz: .19% 1752 MHz: .37% 1980 MHz: .16% 2208 MHz: .23% 2448 MHz: .28% 2676 MHz: .13% 2904 MHz: .05% 3036 MHz: .01% 3132 MHz: .08% 3168 MHz:   0% 3228 MHz: .20%)

P1-Cluster Power: 44 mW
P1-Cluster HW active frequency: 758 MHz
P1-Cluster HW active residency:   2.81% (600 MHz:  85% 828 MHz: .35% 1056 MHz: 4.6% 1296 MHz: 3.5% 1524 MHz: .93% 1752 MHz: 1.6% 1980 MHz: .88% 2208 MHz: .40% 2448 MHz: 1.0% 2676 MHz: .74% 2904 MHz: .12% 3036 MHz: .05% 3132 MHz: .13% 3168 MHz:   0% 3228 MHz: .95%)
P1-Cluster idle residency:  97.19%
P1-Cluster instructions retired: 5.97208e+08
P1-Cluster instructions per clock: 1.60433
CPU 6 frequency: 1844 MHz
CPU 6 idle residency:  97.59%
CPU 6 active residency:   2.41% (600 MHz: .08% 828 MHz: .01% 1056 MHz: .44% 1296 MHz: .31% 1524 MHz: .10% 1752 MHz: .32% 1980 MHz: .37% 2208 MHz: .15% 2448 MHz: .24% 2676 MHz: .11% 2904 MHz: .08% 3036 MHz:   0% 3132 MHz: .05% 3168 MHz:   0% 3228 MHz: .14%)
CPU 7 frequency: 2020 MHz
CPU 7 idle residency:  99.10%
CPU 7 active residency:   0.90% (600 MHz: .02% 828 MHz: .00% 1056 MHz: .14% 1296 MHz: .09% 1524 MHz: .05% 1752 MHz: .12% 1980 MHz: .10% 2208 MHz: .07% 2448 MHz: .11% 2676 MHz: .01% 2904 MHz: .01% 3036 MHz:   0% 3132 MHz: .03% 3168 MHz:   0% 3228 MHz: .15%)
CPU 8 frequency: 2056 MHz
CPU 8 idle residency:  99.67%
CPU 8 active residency:   0.33% (600 MHz: .01% 828 MHz: .00% 1056 MHz: .03% 1296 MHz: .03% 1524 MHz: .01% 1752 MHz: .03% 1980 MHz: .08% 2208 MHz: .01% 2448 MHz: .07% 2676 MHz: .01% 2904 MHz: .01% 3036 MHz:   0% 3132 MHz: .02% 3168 MHz:   0% 3228 MHz: .02%)
CPU 9 frequency: 2176 MHz
CPU 9 idle residency:  99.67%
CPU 9 active residency:   0.33% (600 MHz: .00% 828 MHz:   0% 1056 MHz: .02% 1296 MHz: .03% 1524 MHz: .00% 1752 MHz: .05% 1980 MHz: .07% 2208 MHz: .02% 2448 MHz: .06% 2676 MHz: .00% 2904 MHz: .00% 3036 MHz:   0% 3132 MHz: .02% 3168 MHz:   0% 3228 MHz: .05%)

System instructions retired: 9.85377e+09
System instructions per clock: 1.08824
ANE Power: 0 mW
DRAM Power: 364 mW
CPU Power: 394 mW
GPU Power: 32 mW
Package Power: 1402 mW
```

BitesPotatoBacks, 2022-07-01:
@TianyiShi2001 Ah, `powermetrics`. It is a viable tool, but requires `sudo`. Could be fine for the project to implement, but a password prompt would most likely be needed, and how the user password would be stored would be a security thing that would need sorted...

But! Retrieving CPU frequency statistics is actually doable using Objective-C. I've done it for my own project.

If done in straight code, you would need to use a private framework called `IOReport`. You can add it in an Xcode project under Build Phases>Link Bianry with Libraries. You'll then need to call some extrern funcs ([From this header](https://github.com/samdmarshall/OSXPrivateSDK/blob/master/PrivateSDK10.10.sparse.sdk/usr/local/include/IOReport.h)). Then, you'll have to make dual samples between an interval (minimum 1ms, 1s recommended): 
```C
// data we need
CFMutableDictionaryRef subbedchannels = NULL;
CFMutableDictionaryRef channels = IOReportCopyChannelsInGroup(@"CPU Stats", 0, 0, 0, 0));
IOReportSubscriptionRef sub =IOReportCreateSubscription(NULL, cpuchannels, &subbedchannels, 0, 0));

CFDictionaryRef  samples_a = IOReportCreateSamples(sub, subbedchannels, NULL)); // first samp
[NSThread sleepForTimeInterval: (1 * 1e-3)]; // 1ms
CFDictionaryRef  samples_b = IOReportCreateSamples(sub, subbedchannels, NULL)); // last Samp
```
Then you'll need to find the delta with
```C
CFDictionaryRef samples = IOReportCreateSamplesDelta(samples_a, samples_b, NULL));
```
Then you'll have to iterate through `samples` and pull the data you want (i.e. `CPU Complex Performance States` -> `ECPU`), once you do, you'll have to do some tricky formatting to find CPU Cluster pstate residencies (as well pull additional data from pmgr in `IORegistry` so you actually know the nominal frequencies tied to each of those pstates), and do some more fun stuff to find your frequency. This is the same thing `powermetrics` does, you just won't need `sudo` when you're finished. I know this because I've spent more time than I'd like to reverse engineering `powermetrics`.

The code is actually very minimal and simple, just to long to put here.  This comment is all ready probably way too long (sorry about that 😅 ) but thought I'd share the alternative option to `powermetrics`.

@exelban If you are to look into implementing this request and want more info on the objective-c method (👆), I can put it in a gist so you can see the full code.

exelban, 2022-07-02:
@BitesPotatoBacks hi. Yes, I checked this way to obtain the CPU frequency. And it does not work for me. I have tested it on M1 and M1 Max and also intel. For all of these silicons, the value was too strange. And comparing to the IPG on intel CPU it was a totally wrong value.

TL.DR: it's not possible to obtain a real CPU frequency for Apple Silicon (without root).

closed 2022-07-02 by exelban

BitesPotatoBacks, 2022-07-02:
@exelban I heavily apologize for commenting now with the issue being closed, but I do believe you can acquire frequencies on apple silicon without root, as I _have_ managed to do it.

Could [you run my script](https://github.com/BitesPotatoBacks/MxSocPowerBuddy) and take a look at the output on your Apple M1 systems, compared to Powermetrics? (If you do, try running the command string listed in the "Example" of the README) I only wish to confirm whether this trick works, because from my experience, it does seem to and I would love for Stats app to implement it.

Also, when you wrote your code, did you ever pull a value called `voltage-states1-sram` from Ioregistry pmgr? You'll need the values from that property to find the freq, so That may have been the issue...