exelban/stats #986: [Feature Request] Add button to close all stack view open for specific option

kevinm6, 2022-07-02:
I think it could be nice add a single button (maybe a disclosure next to the popup title) that close/open all the chart (StackView) for a specific module (especially “Sensors”).

exelban, 2022-07-04:
Hi. Thanks for the feedback.
I don't see a good reason for this button. It's easy to close 10+ charts as open them. And I don't want to invent a new mechanism for that type of button. I just don't see any reason for that.

closed 2022-07-04 by exelban

kevinm6, 2022-07-04:
Ok, I get what you mean.
I was just thinking that (like tree folders) use a disclosure to toggle all the stack view could be useful, even because the are not reset to hidden by default when showing again the popup... and since, unlike iStatMenus, graph appears when you click instead of showing a popover with it.

So, you don't accept even a PR, in case?

Thank you for the reply and the amazing work on this app, btw.

exelban, 2022-07-05:
Hi.
I take a look at how it can be implemented. It's possible, but it requires too much work for such a small feature.

The problems related to this feature:
- it cannot be just a button "hide all charts" because it must be visible only when at least 1 chart is opened
- ok, it can be a toggle button that will be visible all time but open/close all charts depends... and depends on what? For example, if you have opened one chart, this button must open all charts or close one opened? Every user will have his own opinion on that point.
- it must be a separate button for each kind of sensors