exelban/stats #987: Fan still shown when it is switched off

Omoeba, 2022-07-03:
**Describe the bug**
As shown in the following screenshots, "Fan #0" is still shown in the sensors dropdown even though it is switched off in the settings.
[image]
[image]

**Details:**
 - Device: MacBook Pro 13" (M1, 2020)
 - macOS: 12.4.0
 - Application version: 2.7.23

exelban, 2022-07-03:
Yes, because this option removes it from the menu bar, not from the popup view.

closed 2022-07-03 by exelban

Omoeba, 2022-07-03:
Thank you for clarifying.