exelban/stats #990: CPU widget does not show on menu bar

pr-o, 2022-07-04:
Hi, 

first off, kudos to you and all this great work 👏 .

I ran across an issue where CPU widgets do not show on menu bar.

I've tried turning the **switch** on and off restarting the **app** and yet the error persists.

FYI, using M1 MacBook Pro (16-inch, 2021).

I suppose this information will not suffice enough to debug this issue, so please let me know if you need any other information.

[image]

exelban, 2022-07-04:
Hi. It's an Apple bug. The widget is not visible because of the MacBook screen design. You need to remove some elements from the menu bar to see another one.

closed 2022-07-04 by exelban

pr-o, 2022-07-04:
Aha, because of overflow? I see. Thanks!

earthsound, 2022-08-18:
> Hi. It's an Apple bug. The widget is not visible because of the MacBook screen design. You need to remove some elements from the menu bar to see another one.

Will "[Scale to fit below built-in camera](https://support.apple.com/en-us/HT212842)" help with this?