exelban/stats #994: Command line tool?

ehemmete, 2022-07-08:
It would be very helpful to be able to take advantage of all the work done on Stats in a command line tool.  Specifically, I'm looking for getting temperature and fan speed as text output, but a general way of asking for any of the available data would be awesome.  I've played around with trying to wire up the smc cli tool and the name resolution bits, but it seems I'm too much of a novice.
Something like /Applications/Stats.app/Contents/Stats-cli -t -f for temp and fans.  Then output text, one name/value pair per line.  To get really fancy some kind of json output could be cool, but just parsable text would be great.

exelban, 2022-07-08:
Hi.
There is one.

`/Applications/Stats.app/Contents/Resources/smc list` - all sensors
`/Applications/Stats.app/Contents/Resources/smc fans` - fans info

`/Applications/Stats.app/Contents/Resources/smc help` - all commands

closed 2022-07-08 by exelban

ehemmete, 2022-07-08:
Yes, but that doesn't map to nice names like the GUI does.  I'm looking for a way to get the useful name from things like [Ts1a]     35.38265609741211

exelban, 2022-07-09:
Nope, there is no way to do it. And I have no plan to integrate the dictionary into the `SMC` tool.

9840380, 2023-07-18:
Any way to get cpu % Usage from cli?