exelban/stats #1002: [Feature Request] Auto switch fan controller back to automatic mode when the macbook lid is closed / macbook is slept

ngshijun, 2022-07-15:
Hello, 

I'd like to suggest to a new feature which allows the fan controller to be switch back from manual to automatic mode once the lid is closed.

It has happened to me many times where I leave the fan to run at 3k+ rpm after some heavy task, and only realised it after few hours when I come back to my MacBook.

exelban, 2022-07-17:
Hi.
I'm pretty sure the fan will be restored to the auto after waking up. I will recheck that. But mainly because of that, there is an option `Save the fan speed`.

ngshijun, 2022-08-12:
Hi, after several times trying, the issue still persist. If I close my macbook lid without changing the fan speed back to auto, the fan will still spins at the fan speed I set even when the lid is closed.

exelban, 2022-08-16:
I'm not sure it's only on apple silicon (previously it does not work like this on intel based macs) the mac control fans are even in sleep mode. So I will add a fan reset before going to sleep mode. And it will be restored if the `Save the fan speed` option is enabled.

ngshijun, 2022-08-16:
Thank you very much 🥰

closed 2022-08-21 by exelban