exelban/stats #1008: Dark mode for the main UI

ryanccn, 2022-07-23:
Hello,

It would be great if you could add dark mode support to the main config screen! Currently only the dashboard and the settings views have dark mode; all the others have a light background.

exelban, 2022-07-24:
Hi. Thanks for the feedback.
Currently, there is no support for modes. Dark colors do not mean that it's a dark mode.

I will leave this issue open, maybe someone wants to help with an app redesign supporting the modes.

ryanccn, 2022-09-04:
> Dark colors do not mean that it's a dark mode.

I'm not sure if my understanding is correct, but does this mean that the two pages that do have color modes are provided by system UI toolkits, and the other pages are self designed and thus do not have color modes?

closed 2022-12-04 by exelban