exelban/stats #1009: Temperature of diode and virtual is always (melting) high

zangobot, 2022-07-24:
**Describe the bug**
On a Macbook Air 2013, the CPU diode virtual and filtered is always at 129°.
But the overall temperature of the computer is not.
I am attaching a screenshot of the output.

**Details:**
 - Device: Macbook Air 2013
 - macOS: 11.6.7
 - Application version: 2.7.27
 
 
[image]

zangobot, 2022-07-24:
Updated to 11.6.8 and the issue disappeared.

closed 2022-07-24 by zangobot

e1z0, 2023-02-08:
iMac Late 2014, macOS Ventura

[image]