exelban/stats #1012: M2 CPU and GPU temperature sensors support

mooons, 2022-07-26:
**Describe the bug**

M2 sensors support is incomplete at the moment. So I've spent some time and have been able to figure out which M2 sensor key corresponds to which component. Partially by generating some load and comparing `/Applications/Stats.app/Contents/Resources/smc list -t` output with other existing temperture monitoring solutions.

Some refactoring might be required to support both M1-series and M2 sensors at the same time, since the same sensors keys are reused for different parts in different models. For example, `Tp05` is `CPU performance core 2` for M1 Pro / M1 Max, but the same key is `CPU efficiency core 1` for M2.

```
    // M2
    Sensor(key: "Tp05", name: "CPU efficiency core 1", group: .CPU, type: .temperature, average: true),
    Sensor(key: "Tp0D", name: "CPU efficiency core 2", group: .CPU, type: .temperature, average: true),
    Sensor(key: "Tp0j", name: "CPU efficiency core 3", group: .CPU, type: .temperature, average: true),
    Sensor(key: "Tp0r", name: "CPU efficiency core 4", group: .CPU, type: .temperature, average: true),

    Sensor(key: "Tp01", name: "CPU performance core 1", group: .CPU, type: .temperature, average: true),
    Sensor(key: "Tp09", name: "CPU performance core 2", group: .CPU, type: .temperature, average: true),
    Sensor(key: "Tp0f", name: "CPU performance core 3", group: .CPU, type: .temperature, average: true),
    Sensor(key: "Tp0n", name: "CPU performance core 4", group: .CPU, type: .temperature, average: true),

    Sensor(key: "Tg0f", name: "GPU 1", group: .GPU, type: .temperature, average: true),
    Sensor(key: "Tg0n", name: "GPU 2", group: .GPU, type: .temperature, average: true),
```

A cleaner solution I can think of is to group sensors by Mac models.

fyi I have posted the version I'm using right now to my fork's dev branch, which works for me (but not for M1/M1 Pro/M1 Max/M1 Ultra anymore since I have commented out those conflicting sensor keys):

https://github.com/mooons/stats/commits/dev-m2

Hope this would help. :)

**Details:**
 - Device: MacBook Air (M2, 2022)
 - macOS: 12.5 (21G72)
 - Application version: 2.7.27

exelban, 2022-07-27:
Hi.
Thanks for your feedback. It's interesting. It looks like Apple will make breaking changes on every silicon revision(
Could I ask how you detect these sensors?

mooons, 2022-07-27:
Hi @exelban , sure thing. As I realize Stats is missing a bunch of new sensors key for M2, I went and see if others have already added such sensors.

I found TG Pro (2.71) and Macs Fan Control (1.5.14 beta) does support M2's sensors. So I generated single core workloads and let one of the core temperature spike to ~70 C, then check `smc list -t | grep "Tp"` result (using `watch` commmand and taking a lot of screenshots) with theirs. After some back and forth changing sensor keys, I am able to match my dev branches' output to TG Pro's (2.71) and Macs Fan Control's (1.5.14 beta). For example:

```bash
$ /Applications/Stats.app/Contents/Resources/smc list -t | grep "Tp"
[Tp00]     46.29172897338867
[Tp01]     52.391727447509766
[Tp02]     66.7281265258789
[Tp04]     46.477783203125
[Tp05]     51.564781188964844
[Tp06]     59.33412551879883
[Tp08]     46.657047271728516
[Tp09]     52.75704574584961
[Tp0A]     58.400001525878906
[Tp0C]     46.36359786987305
[Tp0D]     51.450599670410156
[Tp0E]     55.39662551879883
[Tp0a]     46.651611328125
[Tp0b]     52.10161209106445
[Tp0c]     51.25
[Tp0e]     51.31259536743164
[Tp0f]     57.412593841552734
[Tp0g]     59.415626525878906
[Tp0i]     52.513763427734375
[Tp0j]     57.60076141357422
[Tp0k]     58.97475051879883
[Tp0m]     66.79479217529297
[Tp0n]     72.89479064941406
[Tp0o]     75.3531265258789
[Tp0q]     59.34077835083008
[Tp0r]     64.42778015136719
[Tp0s]     66.19349670410156
```

GPU is much easier as there are only 6 sensors to guess from. `GFXBench Metal` generates enough load for M2 Air's GPU cluster to easily hit 80-90 C.

```bash
$ /Applications/Stats.app/Contents/Resources/smc list -t | grep "Tg"
[Tg0e]     78.25640106201172
[Tg0f]     82.8644027709961
[Tg0m]     78.75232696533203
[Tg0n]     83.3603286743164
[Tg0q]     72.22441101074219
[Tg0r]     76.83241271972656
```

I had tried https://github.com/fermion-star/apple_sensors at one point but it can't seem to read those new temperature sensors correctly. If there is any better way to do this pls do let me know (apart from guessing and matching). :)

[image]

exelban, 2022-07-28:
Thanks for sharing.
The best way I know it's guessing and matching(

Will try to implement these keys to the app. Need to detect m2 before.

jamesdbailey, 2022-08-11:
It's interesting that the IOKit HID AppleSMC keys have changed between the M1 and the M2 as well. I've been trying to understand the differences but it is pretty obscure for what the M2 keys represent while the M1 was pretty easy to figure out what was a temperature sensor for an efficiency core vs. a performance core vs. a GPU core based on the names.

jamesdbailey, 2022-08-16:
I have a simple repository with code to detect the CPU Brand, the number of performance levels and the number of logical cpu cores per performance level.

Code is here: [https://github.com/jamesdbailey/cpuinfo.git](url)

exelban, 2022-08-26:
Hi. I don't have an m2, could someone check if the sensors are correct?
[Stats.dmg.zip](https://github.com/exelban/stats/files/9436215/Stats.dmg.zip)

closed 2022-08-28 by exelban

mooons, 2022-10-14:
> Hi. I don't have an m2, could someone check if the sensors are correct? [Stats.dmg.zip](https://github.com/exelban/stats/files/9436215/Stats.dmg.zip)

Hi @exelban . I can confirm the latest stats (2.7.38) shows E/P cores temp as expected on my M2. Thanks for working on this!