exelban/stats #1016: Can't change menu bar icons

andrewc1989, 2022-07-30:
Can't change the menu bar icons in Mac OS Monterey 12.5

andrewc1989, 2022-07-30:
(I'm using an M2 Air if that matters)

exelban, 2022-07-31:
I suppose you are talking about widgets. Do you try to enable another widget?

andrewc1989, 2022-07-31:
> Owner

Sorry, yes I mean the widgets. I can't change any, they're all stuck on the default setting.

exelban, 2022-07-31:
What do you mean by `they're all stuck`?

exelban, 2022-07-31:
Can you enable another widget?

andrewc1989, 2022-07-31:
> What do you mean by `they're all stuck`?
[image]

andrewc1989, 2022-07-31:
> Can you enable another widget?

No
[image]
.

exelban, 2022-07-31:
Yeap, and now move the widgets you want to enable from the right to the selector's left part.

closed 2022-07-31 by exelban

andrewc1989, 2022-07-31:
Sorry, first time using. Not very intuitive. Thanks for explaining.