exelban/stats #1017: Not showing external disk activity

egrieco, 2022-07-31:
I'm running an `rsync` of an old backup which is currently transferring movie files. Stats is not showing any drive activity on the external hard drives. They are connected through a USB hub.

System Software Overview:

	System Version: macOS 11.6 (20G165)
	Kernel Version: Darwin 20.6.0
	Boot Mode: Normal
	Secure Virtual Memory: Enabled
	System Integrity Protection: Enabled

What other information is needed to troubleshoot this?

I don't have time to dig through the source right now, I'm leaving this here as a reminder to myself.

exelban, 2022-08-01:
Hi. Thanks for the feedback.
The activity data of the disk depends on the disk drivers. If disk reports the io via driver to the system, the Stats could fetch that and show. If not, I can't do anything with that(

closed 2022-08-01 by exelban

egrieco, 2022-08-03:
That makes sense.

I was using [iStat Menus](https://bjango.com/mac/istatmenus/) prior to switching to Stats and I believe that they were able to display external disk transfers. I wonder what kind of shenanigans they are using to do so.